muki01/OBD2_K-line_Reader #5: could not convert 'WiFi.WiFiClass::config(IPAddress(STA_ip), IPAddress(STA_gateway), IPAddress(STA_subnet))' from 'void' to 'bool'

prai44, 2024-04-18:
C:\Users\Suprin\Documents\OBD2_K-line_Reader-main\ESP32\WEB_SERVER.ino: In function 'void initWiFi()':
WEB_SERVER:17:20: error: could not convert 'WiFi.WiFiClass::config(IPAddress(STA_ip), IPAddress(STA_gateway), IPAddress(STA_subnet))' from 'void' to 'bool'
     if (WiFi.config(STA_ip, STA_gateway, STA_subnet));
         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
WEB_SERVER:19:8: error: 'class WiFiClass' has no member named 'mode'
   WiFi.mode(WIFI_STA);
        ^~~~
WEB_SERVER:19:13: error: 'WIFI_STA' was not declared in this scope
   WiFi.mode(WIFI_STA);
             ^~~~~~~~
WEB_SERVER:20:8: error: 'class WiFiClass' has no member named 'setTxPower'
   WiFi.setTxPower(WIFI_POWER_2dBm);
        ^~~~~~~~~~
WEB_SERVER:20:19: error: 'WIFI_POWER_2dBm' was not declared in this scope
   WiFi.setTxPower(WIFI_POWER_2dBm);
                   ^~~~~~~~~~~~~~~
WEB_SERVER:21:28: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   WiFi.begin(STA_ssid.c_str(), STA_password.c_str());
              ~~~~~~~~~~~~~~^~
In file included from C:\Users\Suprin\Documents\OBD2_K-line_Reader-main\ESP32\ESP32.ino:1:
C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:79:21: note:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)'
     int begin(char* ssid, const char *passphrase);
               ~~~~~~^~~~
WEB_SERVER:30:10: error: 'class WiFiClass' has no member named 'mode'
     WiFi.mode(WIFI_AP);
          ^~~~
WEB_SERVER:30:15: error: 'WIFI_AP' was not declared in this scope
     WiFi.mode(WIFI_AP);
               ^~~~~~~
WEB_SERVER:31:10: error: 'class WiFiClass' has no member named 'softAP'
     WiFi.softAP(AP_ssid, AP_password);
          ^~~~~~
exit status 1
could not convert 'WiFi.WiFiClass::config(IPAddress(STA_ip), IPAddress(STA_gateway), IPAddress(STA_subnet))' from 'void' to 'bool'

muki01, 2024-04-18:
Which board are you using ?

muki01, 2024-04-18:
> C:\Users\Suprin\Documents\OBD2_K-line_Reader-main\ESP32\WEB_SERVER.ino: In function 'void initWiFi()': WEB_SERVER:17:20: error: could not convert 'WiFi.WiFiClass::config(IPAddress(STA_ip), [...]

Which board are you using ?

prai44, 2024-04-18:
esp wroom 32

prai44, 2024-04-18:
[image]

muki01, 2024-04-18:
> esp wroom 32

Check the version of the board from board manager and if there is a new version update.

prai44, 2024-04-18:
> > esp wroom 32
> 
> Check the version of the board from board manager and if there is a new version update.

2.0.15 board manger esp32 dev module

muki01, 2024-04-18:
1. Which version of Arduino IDE are you using?
2. Are you sure you have selected ESP32 as the board type? If yes, which ESP32 did you choose in the IDE?

prai44, 2024-04-18:
> 1. Which version of Arduino IDE are you using?
> 2. Are you sure you have selected ESP32 as the board type? If yes, which ESP32 did you choose in the IDE?

ide 1.8.19
esp32 dev module 
board manger 2.0.15

muki01, 2024-04-18:
I test with the same stuff and it works for me. I do not know. If I find a solution, I will write to you. You can test by reinstalling the Arduino IDE and the esp32 board.

muki01, 2024-04-18:
If you have another library for wifi in this directory 'C:\Users\USERNAME\Documents\Arduino\libraries' , folder like "Wifi" rename it like "Wifi1" or something else and test.

muki01, 2024-04-18:
I tested in a virtual machine from scratch
1. I installed Arduino IDE 1.8.19
2. I added a link for ESP32 in Arduino IDE. https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
3. I installed ESP32 from the board manager
4. I installed the ESPAsyncWebServer library from the library manager. It asked if i wanted to install AsyncTCP library as well and i clicked "install all"
5. I installed the ArduinoJson library.
6. I clicked Compile button and it didn't give me any error.

Here is all Librarys I installed  [Libraries.zip](https://github.com/muki01/OBD2_K-line_Reader/files/15027751/Libraries.zip)

prai44, 2024-04-18:
> 2\. https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

WEB_SERVER:156:9: error: 'void ws()' redeclared as different kind of symbol
 void ws() {
         ^
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino:9:16: note: previous declaration 'AsyncWebSocket ws'
 AsyncWebSocket ws("/ws");
                ^~
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino: In function 'void setup()':
ESP32:42:3: error: 'initWebSocket' was not declared in this scope
   initWebSocket();
   ^~~~~~~~~~~~~
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino:42:3: note: suggested alternative: 'initWebServer'
   initWebSocket();
   ^~~~~~~~~~~~~
   initWebServer
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino: In function 'void loop()':
ESP32:65:5: error: 'wsSend' was not declared in this scope
     wsSend();
     ^~~~~~
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino:65:5: note: suggested alternative: 'wscanf'
     wsSend();
     ^~~~~~
     wscanf
E:\OBD2_K-line_Reader-main\ESP32\K_Line.ino: In function 'void read_K()':
K_Line:40:3: error: 'wsSend' was not declared in this scope
   wsSend();
   ^~~~~~
E:\OBD2_K-line_Reader-main\ESP32\K_Line.ino:40:3: note: suggested alternative: 'wscanf'
   wsSend();
   ^~~~~~
   wscanf
E:\OBD2_K-line_Reader-main\ESP32\WEB_SERVER.ino: At global scope:
WEB_SERVER:156:9: error: 'void ws()' redeclared as different kind of symbol
 void ws() {
         ^
E:\OBD2_K-line_Reader-main\ESP32\ESP32.ino:9:16: note: previous declaration 'AsyncWebSocket ws'
 AsyncWebSocket ws("/ws");
                ^~
E:\OBD2_K-line_Reader-main\ESP32\WEB_SERVER.ino: In function 'void ws()':
WEB_SERVER:157:3: error: 'webSocket' was not declared in this scope
   webSocket.loop();
   ^~~~~~~~~
E:\OBD2_K-line_Reader-main\ESP32\WEB_SERVER.ino:157:3: note: suggested alternative: 'eBlocked'
   webSocket.loop();
   ^~~~~~~~~
   eBlocked
exit status 1
'void ws()' redeclared as different kind of symbol

muki01, 2024-04-18:
Download my Last commits. Open ESP32.ino file. After that you need to see 4 pages like this:
[image]

If you see these pages, then there is no problem.
Press the compile button without changing the code.

prai44, 2024-04-18:
code compile but not so starting ip
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
ets Jul 29 2019 12:21:46

muki01, 2024-04-18:
Does it work ?

prai44, 2024-04-18:
peramiter set but connect phone hotspot

const char* AP_ssid = "OBD2 Master";
const char* AP_password = "12345678";

IPAddress AP_ip(192, 168, 53, 60);
IPAddress AP_subnet(255, 255, 255, 0);
IPAddress AP_gateway(192, 168, 53, 52);

prai44, 2024-04-18:
conection not stables i have try ssd id rename

prai44, 2024-04-18:
not connect
String STA_ssid = "prince";
String STA_password = "11111111";
String IP_address = "";
String SubnetMask = "";
String Gateway = "";
String protocol = "";

prai44, 2024-04-18:
any desk access in my laptop

muki01, 2024-04-18:
1. If you want to change the IP address, you need to enable this line // if (WiFi.softAPConfig(AP_ip, AP_gateway, AP_subnet));
2. If the connection is not stable, you can increase the wifi power (dBm) in this line WiFi.setTxPower(WIFI_POWER_2dBm); or delete for default value. I scaled back because I'm using an ESP32 C3 Mini, it overheats a lot.

muki01, 2024-04-18:
Do not change settings.conf file. 
ESP32 first creates an access point. If you want to connect to your WiFi connection, you can do it from the web server in settings menu.

muki01, 2024-04-18:
If you explain better what you want, I can help you better.

prai44, 2024-04-18:
I am not able to make the network stable, I have tried all the things which you have told, can you understand me by remote accessing my laptop or can you tell me the step to step network configuration of the code?

muki01, 2024-04-18:
Have you uploaded the website to SPIFFS?
If you have uploaded, follow these steps:

1. Don't change my code and upload it to your ESP32
2. ESP32 Creates a WiFi (access point) which you can see on the wifi page of your device.
3. You need to connect to this WiFi (access point) called "OBD2 Master". Тhe password is "12345678"
4. Then you need to open "192.168.4.1" in your browser
5. You should see this page:
[image]

6. Just ,If you want to connect your ESP32 to your home WiFi, you can do so in the settings page.
[image]

prai44, 2024-04-18:
[image]

prai44, 2024-04-18:
not show wifi ssid

muki01, 2024-04-18:
You dont need Serial Monitor.

prai44, 2024-04-18:
The code has been run but ssd name is obb2 master ssd is not showing in my phone.

muki01, 2024-04-18:
The code does not print the SSID of the serial monitor.
This is how you should see in your wifis.
[image]

prai44, 2024-04-18:
not show i have allready tried

muki01, 2024-04-18:
Delete this line and test WiFi.setTxPower(WIFI_POWER_2dBm);

prai44, 2024-04-18:
allready check it

muki01, 2024-04-18:
Have you uploaded to SPIFFS??

prai44, 2024-04-18:
yes

muki01, 2024-04-18:
Give me AnyDesk access

prai44, 2024-04-18:
Hi,

I would like to invite you to an AnyDesk session. My ID is 1 088 227 428.

Get AnyDesk at:

https://anydesk.com/

prai44, 2024-04-19:
ANY SOLUTION

prai44, 2024-04-23:
Not Working Tried this code many times but couldn't find solution please resolve solution

prai44, 2024-04-23:
[image]

muki01, 2024-05-04:
Hello. I think I found your problem. I tried with ESP32 Devkit and application gave the same error. The problem is that you shouldn't use pin 8. In the code, pin 8 is used to control an LED. Change it to some other number and try. Before, it didn't give me an error because I was using an ESP32 C3 model. ESP32 c3 and ESP32 Wroom pins are different.

prai44, 2024-05-09:
Working fine Can data saving features be added to it?

prai44, 2024-05-10:
[image]
Pin not connected  obd Kline adapter

muki01, 2024-05-10:
What does this mean.  Do you have any problem?

prai44, 2024-05-10:
Not connecting to car please send pin to diagram ka line

prai44, 2024-05-10:
[image]
[image]

prai44, 2024-05-10:
[image]

muki01, 2024-05-10:
Did you do the communication scheme I shared?

prai44, 2024-05-10:
Yes

muki01, 2024-05-10:
Which scheme did you make?

muki01, 2024-05-10:
And why you are using pin 32 and pin 33 ? You need to use UART Pins.

prai44, 2024-05-10:
Please send me complete image connecting pin with espboard

muki01, 2024-05-10:
My ESP32 is different than yours so you should not use as mine. There is nothing complicated, you just need to use the UART pins of the microcontroller. On ESP32 devkit the UART Pins are GPIO1 and GPIO3 (RX0 and TX0). Also in the code you need to set K_line_TX to 1 and K_line_RX to 3. And that's it.

redave, 2024-10-28:
i like this project and using on my car all work and good job my friend. would you add CAN with TJA1050 with this project code because on VW need CAN protocol for this. thank you

muki01, 2024-10-28:
> i like this project and using on my car all work and good job my friend. would you add CAN with TJA1050 with this project code because on VW need CAN protocol for this. thank you

That's exactly my goal right now. I will make version 2 of this project and it will include the CAN protocols and make a good PCB. If you have more questions or want to write to me, create a new Issue. 🙂