muki01/OBD2_K-line_Reader #7: ONE ERROR

toprakb66, 2024-09-14:
"Hello my friend, I hope you're doing well. I made this project and connected everything according to the schematic you showed using Arduino Nano. I also added the code and connected it to the car, but I can't establish communication with the OBD no matter what I do. There's no communication, and I'm about to lose my mind. Can you give me some ideas?"

toprakb66, 2024-09-14:
"Initialising... The serial monitor keeps saying this."

muki01, 2024-09-15:
1. Which scheme are you using?

 2. Which pins are you using with Arduino nano?

 3. Is there pin 7 on your car's OBD2 connector?

toprakb66, 2024-09-15:
"I am using pins 7 and 8, yes, my car's OBD has a pin 7. I am using the Basic Code, the one with the .ino file."

toprakb66, 2024-09-15:
"I was able to get a response when I connected the OBD2 device using the 'ISO9141' and 'ISO14230_Slow' protocols. There is no issue with the car."

toprakb66, 2024-09-15:
"I'm stuck in this loop with the messages: 'Initialising... Received Data: 55 8 8 Received Data: F7'. What could be causing this?"

toprakb66, 2024-09-15:
"I’m receiving the following responses when using the protocol String protocol = "ISO14230_Fast";:

Live Data: Writing Data Received Data: C2 33 F1 1 D F4 Speed: 0 Writing Data Received Data: C2 33 F1 1 C F3 Engine RPM: 0 Writing Data Received Data: C2 33 F1 1 5 EC Coolant Temp: 0 Writing Data"

toprakb66, 2024-09-15:
Received Data: C2 33 F1 1 5 EC 83 F1 18 7F 1 12 1E 
Coolant Temp: 0
Writing Data
Received Data: C2 33 F1 1 F F6 83 F1 18 7F 1 12 1E 
Intake Air Temp: 0
Writing Data
Received Data: C2 33 F1 1 11 F8 83 F1 18 7F 1 12 1E 
Throttle: 0
Writing Data
Received Data: C2 33 F1 1 E F5 83 F1 18 7F 1 12 1E 
Timing Advance: 0

muki01, 2024-09-15:
> Received Data: C2 33 F1 1 5 EC 83 F1 18 7F 1 12 1E Coolant Temp: 0 Writing Data Received Data: C2 33 F1 1 F F6 83 F1 18 7F 1 12 1E Intake Air Temp: 0 Writing Data Received Data: C2 33 F1 1 11 F8 83 [...]

If you get data like "C2 33 F1 1 5 EC 83 F1 18 7F 1 12 1E", it means your Arduino is connected to car ECU. This request code "C2 33 F1 1 5 EC" is written by Arduino and this "83 F1 18 7F 1 12 1E" is the response of the car, but this response means an error. I think your cars protocol is ISO9141 because you are getting error like this. Can you message me the console output in ISO9141 protocol.

toprakb66, 2024-09-15:
"Of course, buddy. The result with the 9141 protocol is as follows:

Received Data: F7
Initialising...
Received Data: 55 8 8
Received Data: F7
Initialising...
Received Data: 55 8 8
Received Data: F7
Initialising...
Received Data: 55 8 8
Received Data: F7
Initialising...
Received Data: 55 8 8
Received Data: F7
Initialising..."

muki01, 2024-09-15:
Ops my bad your protocol is ISO14230_Slow but i dont know why your car response like this. What you receive here 
```
Serial.println("Suported LiveData: ");
  for (int i = 0; i < 32; i++) {
    Serial.print(supportedLiveData[i]);
    Serial.print(", ");
  }
  Serial.println();
```
after you Arduino is connected to your car?

toprakb66, 2024-09-15:
"Yes, the Arduino is currently connected to my car. Could there be an issue in the code?"

muki01, 2024-09-15:
Wait, I will modify the code and you will try with the modified code.

toprakb66, 2024-09-15:
"My friend, I’m waiting excitedly for your response. I’m so grateful for your help, I’ve been working on this for four days now."

muki01, 2024-09-15:
I think I had a bug in the Basic code. Try this code and send me serial output.
[Basic_Code.zip](https://github.com/user-attachments/files/17007068/Basic_Code.zip)

muki01, 2024-09-15:
Keep the protocol in automatic mode, don't change it and send me serial output.

toprakb66, 2024-09-15:
I'm encountering the following error in my new code:  C:\Users\elato\AppData\Local\Temp\5631c5f7-2852-428a-a332-5a3985e80d62_Basic_Code.zip.d62\Basic_Code\Basic_Code.ino: In function 'void loop()':
C:\Users\elato\AppData\Local\Temp\5631c5f7-2852-428a-a332-5a3985e80d62_Basic_Code.zip.d62\Basic_Code\Basic_Code.ino:38:25: error: 'init_OBD2' was not declared in this scope
     bool init_success = init_OBD2();
                         ^~~~~~~~~
C:\Users\elato\AppData\Local\Temp\5631c5f7-2852-428a-a332-5a3985e80d62_Basic_Code.zip.d62\Basic_Code\Basic_Code.ino:38:25: note: suggested alternative: 'init_OBD'
     bool init_success = init_OBD2();
                         ^~~~~~~~~
                         init_OBD
C:\Users\elato\AppData\Local\Temp\5631c5f7-2852-428a-a332-5a3985e80d62_Basic_Code.zip.d62\Basic_Code\Basic_Code.ino:46:5: error: 'read_K' was not declared in this scope
     read_K();
     ^~~~~~
C:\Users\elato\AppData\Local\Temp\5631c5f7-2852-428a-a332-5a3985e80d62_Basic_Code.zip.d62\Basic_Code\Basic_Code.ino:46:5: note: suggested alternative: 'read_ID'
     read_K();
     ^~~~~~
     read_ID

exit status 1

Compilation error: 'init_OBD2' was not declared in this scope
 It seems like a library error, and I'm currently working on fixing it.

muki01, 2024-09-15:
Unzip all the files into one folder and it will work.

toprakb66, 2024-09-15:
`Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
`

toprakb66, 2024-09-15:
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...
Received Data: 55 8 8 
Your Protocol is ISO9141
Received Data: F7 
Initialising...

toprakb66, 2024-09-15:
"I can only see these kinds of values in Fast mode: Speed: 0
Writing Data
Received Data: C2 33 F1 1 C F3 83 F1 18 7F 1 12 1E
Engine RPM: 0
Writing Data
Received Data: C2 33 F1 1 5 EC 83 F1 18 7F 1 12 1E
Coolant Temp: 0
Live Data:
Writing Data
Received Data: C2 33 F1 1 D F4 83 F1 18 7F 1 12 1E
Speed: 0
Writing Data
Received Data: C2 33 F1 1 C F3 83 F1 18 7F 1 12 1E
Engine RPM: 0
Writing Data
Received Data: C2 33 F1 1 5 EC 83 F1 18 7F 1 12 1E
Coolant Temp: 0
Live Data:"

Let me know if you need further help! (FAST PROTOCOL)

muki01, 2024-09-15:
I don't know why this happens to you. I have tested all the protocols and they worked for me. I guess it's due to the timing because all ECUs send the response at different times. I'll tell you a few things to change and try.

1. Increase the time in line 66 "REQUEST_DELAY = 50;" from 50 to 100 and test.
2. In line 70 "if (resultBuffer[1])" change resultBuffer[1] to resultBuffer[0]

Try these steps separately. Keep the protocol in automatic mode.

muki01, 2024-09-15:
I dont know how you can receive data like this C2 33 F1 1 C F3 83 F1 18 7F 1 12 1E . This means you are initialised in ISO14230_Fast but your protocol is ISO9141. This is very strange.

toprakb66, 2024-09-15:
"Buddy, I translated it and here’s what happened:

Live Data:
Writing Data
Received Data: 68 6A F1 1 D D1
Speed: 0
Writing Data
Initialising...
Received Data: 55 8 8
Your Protocol is ISO9141
Received Data: F7
Init Success !!
Live Data:
Writing Data
Received Data: 68 6A F1 1 D D1
Speed: 0
Writing Data
Received Data: 68 6A F1 1 C D0
Engine RPM: 0"

toprakb66, 2024-09-15:
After the last adjustment, the results turned out like this.

toprakb66, 2024-09-15:
Although a value has been read from the OBD, it's indicating 161 degrees Celsius (based on the conversion from hex C9), which seems to be incorrect. This suggests there's an issue either with how the data is being interpreted or a sensor error. Received Data: 68 6A F1 1 5 C9 
Coolant Temp: 0

toprakb66, 2024-09-15:
That means it worked!

muki01, 2024-09-15:
No, this 68 6A F1 1 5 C9 is the request code, not the response from the ECU.

muki01, 2024-09-15:
Ops my bad not in lines 66 and 70 in lines 99 and 103. if use changed in this line and the response is this, change the time to 40 in line 99. and in line 103 the code should look like this "if (resultBuffer[1]) {"

toprakb66, 2024-09-15:
Unfortunately, the result is the same. Are there any other alternatives?

muki01, 2024-09-16:
Which pins do you have in your car's OBD2 port
  for example in the OBD2 port of my car I have pins 2,4,5,7,8,16

toprakb66, 2024-09-17:
"Yes, my friend, I still need your help. I'm busy with farming, and my workload is intense, so I haven't been able to look into it. Sorry about that

toprakb66, 2024-09-17:
My friend, these pins are occupied: obd2 16, 12, 8, 7, 3, 4.

toprakb66, 2024-09-17:
[image]

toprakb66, 2024-09-17:
Friend, Pin 4 is the chassis ground, and I hadn't connected it before. Now, I'll connect it and try again. Hopefully, this is where the issue lies

muki01, 2024-09-17:
Pin 4 and 5 are ground and I think it doesn't matter which one you use.  Which communication scheme did you use?

toprakb66, 2024-09-17:
[image]

toprakb66, 2024-09-17:
[image]

toprakb66, 2024-09-17:
I've now connected pins 4 and 5, and I'm heading to the car to test it

muki01, 2024-09-17:
I don't know why you can't connect to the ECU. Your ECU sends the correct data at the beginning of the initialization 55 8 8 is sent by your ECU and this says that your communication protocol is İSO9141, then you should send F7 and your ECU should respond with another byte and the communication will be OK, but  your ECU is not responding. I think this could happen from the timing in the code or something else. I have tested with a car that had this protocol and it worked perfectly.

muki01, 2024-09-17:
You can try changing the time values in line 97 and 99. Increase, decrease and test. I think that might be the problem. And I have a question. Does your car have an LPG system ?

toprakb66, 2024-09-17:
Sorry, my friend, I don't have an LPG system; my car is a diesel Kia Sorento 2006.

toprakb66, 2024-09-17:
I'm heading back to the car to try again. I believe we can solve this tonight

muki01, 2024-09-17:
I ask because some cars with an LPG system are difficult to connect, because for the LPG system there must be emulators and these emulators are connected to the same protocol and the emulator is constantly sending codes and the codes are mixed with the ECU codes.

muki01, 2024-09-17:
Can you tell me the engine and horsepower of your car?

toprakb66, 2024-09-17:
140 hp, 2.5-liter engine

muki01, 2024-09-17:
Look, I found this on the internet.
[image]

Your protocol is ISO9141, I say, but I don't know why it doesn't connect.

toprakb66, 2024-09-17:
When I connected the Wi-Fi OBD device, I also saw a transmission control unit. I think I’ll connect the Wi-Fi device again and reach out to you."

toprakb66, 2024-09-17:
My friend, there are two control units in the car. One is the ECU for the transmission, and it shows code 0x018 with one data visible. Then there is the ECU for the engine (OBD) with 14 live data points and a code 0x012. There's also the engine manufacturer ECU, which shows 60 errors but no data, and its code is also 0x012

muki01, 2024-09-17:
I understand this well, but before these ECUs there is a layer to connect to the ECU.  First you need to contact to request data from the specific ECU.

toprakb66, 2024-09-17:
[image]

toprakb66, 2024-09-17:
> Bunu iyi anlıyorum, ancak bu ECU'lardan önce ECU'ya bağlanmak için bir katman var. Öncelikle belirli ECU'dan veri istemek için iletişime geçmeniz gerekiyor.
Do you have any idea how we can do this?

muki01, 2024-09-17:
This is exactly what my code is trying to do.  It works fine for me.  Did you change the timings in the code and test?

toprakb66, 2024-09-17:
I tested it by changing lines 97 and 99 one by one.

toprakb66, 2024-09-17:
On my WiFi OBD device, it shows "ISO14230-4 (5baud) start" when selecting the protocol automatically. Is this an important detail?

muki01, 2024-09-17:
> On my WiFi OBD device, it shows "ISO14230-4 (5baud) start" when selecting the protocol automatically. Is this an important detail?

No

muki01, 2024-09-17:
İn line 97 write delay(5); and test. Dont change other in line 99.

muki01, 2024-09-17:
And test with delay(50)

toprakb66, 2024-09-18:
Buddy, my car is out of town, I'll test it as soon as possible.

toprakb66, 2024-09-18:
"Friend, I have another car with K-line, a Lada Samara, and I got these errors as well:  Initialising...
Received Data: FF 0 0 
Writing Data
Received Data: 1 3 F1 81 66 
Initialising...
Received Data: 3F 0 0 
Writing Data

toprakb66, 2024-09-18:
So, it means there’s no problem with the car. Are you running this code with an ESP32, or could there be a problem with my Arduino Nano?"

[95 more replies]