muki01/OBD2_K-line_Reader #8: Not connected

Boak29, 2024-09-30:
Hello friend, i try the basic code with my wemos d1 mini n transistor but did't work

muki01, 2024-09-30:
Hello. Thank you for your feedback. First of all, if you are going to use altserial, you should not change the pin numbers, so it should stay that way. 

`#define K_line_RX 8
#define K_line_TX 9`

muki01, 2024-09-30:
And cant use AltSerial with Wemos

muki01, 2024-09-30:
If you are using Wemos I think there is no way to see the data in the serial monitor because Wemos only has one UART and you have to use it to take data from your car.  The two cannot work simultaneously from one wart, both taking data from the car and printing it out in the serial monitor is not possible.

muki01, 2024-09-30:
you need to delete the serial.print and serial.ptintln codes everywhere, also everything about altserial and declare the pins like this: 

`#define K_Serial Serial
#define K_line_RX 3
#define K_line_TX 1
#define Led 2`

Boak29, 2024-09-30:
> If you are using Wemos I think there is no way to see the data in the serial monitor because Wemos only has one UART and you have to use it to take data from your car.  The two cannot work [...]

So how can i monitor the result?

muki01, 2024-09-30:
In screen or in webserver

muki01, 2024-09-30:
I made a code for ESP8266 you can test this code.  but first you need to upload the web page to the file storage (SPİFFS)

Boak29, 2024-09-30:
thx for the answer, I have no experience in coding displays and web servers, what about 13 & 15, those are rxd2 and txd2, can I use them?

Boak29, 2024-09-30:
> I made a code for ESP8266 you can test this code.  but first you need to upload the web page to the file storage (SPİFFS)

OK, if you don't mind, I'll try it

Boak29, 2024-09-30:
OK, it's as easy as the ESP32 I've tried

Boak29, 2024-09-30:
Thx a lot friend the webserver goes well 
[image]

[image]

Nice jobs

Boak29, 2024-09-30:
tomorrow I'll try to connect to my car

muki01, 2024-09-30:
I'm glad everything is going well so far 😊. Write me back after you test it with your car.

Boak29, 2024-10-01:
hi friend, i have to connect to my car this not work 
[image]

Vehicle wont connect

Boak29, 2024-10-01:
how do i check it?

muki01, 2024-10-01:
1. Which pins are you using in Wemos.
2. Are you sure you have 7th pin in the OBD2 socket.
3. What car are you testing with, model, engine, horse power.

Boak29, 2024-10-01:
On ob2 i have pin 4, 5, 7, 9, 16

muki01, 2024-10-01:
The bad thing is that without a serial monitor it is very difficult to understand the problem.  I had tried a WebSerial but it wasn't very good.  if you still have problem i will make a WebSerial code and you will try it for debuging.

Boak29, 2024-10-01:
[image]

Yes sure i have connected it.

Boak29, 2024-10-01:
I'll try change the protocol setting tommorow, thx for your help

Boak29, 2024-10-01:
> The bad thing is that without a serial monitor it is very difficult to understand the problem.  I had tried a WebSerial but it wasn't very good.  if you still have problem i will make a WebSerial code and you will try it for debuging.

That's why I asked you the first time because I wanted to check this car with a serial monitor, whether my car could be detected or not

Boak29, 2024-10-01:
Thx friend, i have ttgo tdisplay & arduino promini

toprakb66, 2024-10-02:
Did you manage to read the data?

Boak29, 2024-10-05:
Hi friend i have already connected to my car this is the result[image]

I turn on the engine but still 0

Boak29, 2024-10-05:
I use arduino pro mini

muki01, 2024-10-05:
I tried it and I had the same problem.  Change the protocol from Automatic to İSO14230_Fast and it will work.  The problem is when the protocol is on Automatic.  I'll fix it today.

muki01, 2024-10-05:
Just use this code and everything should work for you. 
[Basic_Code.zip](https://github.com/user-attachments/files/17266463/Basic_Code.zip)

Boak29, 2024-10-05:
Ok my friend thx, i'll try tomorrow

Boak29, 2024-10-06:
> Just use this code and everything should work for you. 
> [Basic_Code.zip](https://github.com/user-attachments/files/17266463/Basic_Code.zip)
> 

Hi friend, i use your code & this is the result
[image]

Boak29, 2024-10-06:
still 0

muki01, 2024-10-06:
Hmm, now your car is responding but with an error.  Could you please try the same code with İSO9141 protocol and send me a screenshot again.  To find out if the protocol is correct.

Boak29, 2024-10-06:
FYI i have try elm327 bluetooth to scan my car but just DTC working, the live data not working, the protocol detect iso14230-4

Ok i'll try iso9141 tommorow
Just uncomment this code right?
 String protocol = "ISO9141";

muki01, 2024-10-06:
> FYI i have try elm327 bluetooth to scan my car but just DTC working, the live data not working, the protocol detect iso14230-4
> 
> Ok i'll try iso9141 tommorow
> Just uncomment this code right?
>  String protocol = "ISO9141";

It is strange that it shows only DTCs. If it says the protocol is ISO14230 and shows DTCs, then the protocol should be ISO14230, but you can still try ISO9141 in my code. And yes uncomment line String protocol = "ISO9141"; and comment other lines.

Boak29, 2025-12-30:
hi friend long time no talk with u. i have tried to connected elm327 to my car n get connected with iso kwp fast and with string ATSP5,ATTP5,ATAL,ATIB10,ATSH8110F1,ATFI,1081,3E. i use carbit app. how can i apply it to this code