muki01/OBD2_K-line_Reader #9: Ajout données

chdavgas, 2024-10-03:
Bonjour, 
Votre projet m'intéresse beaucoup !! la version pour l'esp32 et esp8266 est très appréciable. Par contre serait il possible de rajouter les valeur de "SHORT_TERM_FUEL_TRIM_BANK_1" et "LONG_TERM_FUEL_TRIM_BANK_1" à votre programme ? En effet ces valeurs me sont nécessaire pour le réglage à éthanol (e85). En vous remerciant par avance. David

muki01, 2024-10-03:
Bonjour. Merci pour vos commentaires. Je m'excuse si je n'arrive pas à bien m'exprimer car j'isole Google Translate. Pour vous j'ai ajouté ces valeurs et les ai testées, elles fonctionnent pour moi. Vous pouvez télécharger le code ici et l'essayer. J'ai juste une question à te poser, ton code fonctionne-t-il ? Cela a bien fonctionné pour moi, mais d'autres disent que cela n'a pas fonctionné pour eux.

[WebServer_Code.zip](https://github.com/user-attachments/files/17246120/WebServer_Code.zip)

chdavgas, 2024-10-04:
Bonjour, merci pour l'ajout des données. Alors pour répondre à ta question, effectivement j'ai eu du mal a faire fonctionner le code. je l'injectais avec IDE 2.3.2. D 'abord le programme puis les "spiffs". D'après  IDE tout est ok, pas d'erreur. Mais "OBD2 MASTER" n'apparaissait pas dans les réseaux wifi disponible. j'ai désactivais avec //   initSpiffs(); et  readSettings(); et là je pouvais me connecter à OBD2 MASTER mais forcement il ne trouvait pas les pages html. j'en ai conclu un problème avec les spiffs. Apres des recherches sur le web les spiffs ne marche pas avec IDE 2.32. J'ai installé IDE 1.8.19 et tout fonctionne que ca soit sur un ESP32 ou sur un ESP8266. J'ai juste modifié les pins pour ESP32 (RX-> 3, TX-> 1, LED->2, BUZZER -> 21) car d'après ma carte c'est ces pins qu'il faut utiliser. je n'ai pas pu encore l'utiliser sur la voiture car j'attends de recevoir les transistors. Des que je le test, je te tiendrai informer !! Une dernière chose, ton Français est très bon. bonne journée

chdavgas, 2024-10-12:
bonjour, comme convenu je reviens vers toi pour te donner des nouvelles. j'ai procédé à un essai avec les transistors bc547.j ai effectué mes test avec un esp8266-12f monté sur un wemos R1 et avec un esp32-wrom.Dans les deux cas je me connecte sans difficulté à 192.168.4.1. Toutes les pages s'affichent. j 'ai essayé les 3 protocoles.( avec esp32 un pop up s'affiche avant le redémarrage) mais malheureusement toutes les valeurs m'affiche 0. j'ai verifié le montage deux fois. je n'arrive pas a me connecter au calculateur. Ma voiture est une RENAULT Mégane Scenic de 1997 essence 2.0l. les brochres de l'obd sont : 1,3,4,5,7,11,15,16.
4 ET 5 sont GND. j ai bien 12v sur le 16.  voila les informations que je peux te transmettre. n'hesite a a me contacter pour d'autre renseignement ou test.

chdavgas, 2024-10-13:
bonjour, j ai des infos complementaires.
quand j utilise le k-line reader sur le moniteur serie j 'ai initialisation... write data quand il est en automatique. si j utilise le protocole iso 9141 j ai que initialistion...

muki01, 2024-10-13:
Bonjour.  J'ai quelques questions.  Est-il indiqué que vous êtes connecté en vert sur l'écran d'accueil avec un protocole ?  Comment utiliser le moniteur série ?  S'ils utilisent l'esp8266, vous ne devez pas utiliser le moniteur série car il est utilisé pour communiquer avec la voiture.  Les deux choses ne peuvent pas être dans un seul UART.

chdavgas, 2024-10-13:
bonjour, avec la version websever avec un esp32 la communication avec l'adresse 192.168.4.1 fonctionne bien. (deconnection en changeant de page mais reconnections automatique) quand je change de protocole je me reconnecte manuellement mais le protocole est bien pris en compte. par contre je n arrive pas a me connecter au calculateur. j ai essayé la version reader sans altsoftserial car non compatible avec un esp32. Sur le moniteur serie j 'ai initialisation... write data en boucle et initialisation... quand je selectionne le protocole 9141.

muki01, 2024-10-13:
Bonjour. J'ai mis à jour le code de base pour ESP32. Vous pouvez le tester. Utilisez RX pour la broche 16 et TX pour la broche 17 et gardez le protocole en mode automatique. L'écran série doit alors indiquer quel est le protocole de votre voiture et il doit se connecter. S'il ne se connecte pas, modifiez le protocole affiché sur le moniteur série et essayez.

chdavgas, 2024-10-14:
Le calculateur de la voiture est un siemens fenix 5 ( S 103 717 101 B). Les broches 7 et 15 de la prise obd sont reliés sur le calculateur.

muki01, 2024-10-14:
> Bonjour, Merci pour la modification du code. Je recois des data. Dans un premier temps : FC FF F8 puis F8 F7 FE F0 F7. Puis il s initialise de nouveau et m'affiche F8 F7 FE F0 F7. Il m'indique pas le protocole utilisé.

hier j'ai essayé et au début j'obtenais les mêmes données mais pas les données correctes. ensuite j'ai secoué les fils et touché le microcontrôleur avec un prist et c'était bien. Cela peut être à cause de Serial1, je ne sais pas exactement. Sinon, ces données que vous obtenez ne sont pas reçues par la voiture, ce sont les données que l'esp32 envoie pour communiquer. Ces données sont destinées au protocole ISO14230 et devraient ressembler à ceci C1 33 F1 81 66. si vous voyez ces données sur le moniteur série, alors Serial1 fonctionne correctement.

muki01, 2024-11-15:
Bonjour à nouveau. Je n'ai pas eu beaucoup de temps pour examiner cela en détail, mais j'ai maintenant trouvé le problème. le problème vient des broches du microcontrôleur. J'ai changé ces broches en GPIO25 et GPIO26 et tout a bien fonctionné. si vous essayez toujours, vous pouvez essayer ces épingles.

chdavgas, 2024-11-16:
Bonsoir,
Merci de l intérêt que vous portez à mon problème mais de mémoire j avais déjà effectué cette modification pour essayer de résoudre mon problème mais sans résultat. Le problème vient à mon avis du véhicule. Il est de 1997. Il est câblé mais j ai 0 volt sur la pin 7. Pouvez vous me confirmer que cette broche devrait délivrer du 12v ?
Mais j essayerai la semaine des que je rentre chez moi. Bon weekend

muki01, 2024-11-16:
Sur la broche 7, cela indique 0 volt pour moi et cela fonctionne bien, pas de problème.  Testez avec le dernier Basic_Code que j'ai modifié et téléchargez les résultats.