muki01/OBD2_K-line_Reader #16: Volkswagen 2006  ISO 14230-4 KWP (fast baund init 0x8110F1) 10.4 kbaund

vladinen, 2025-03-07:
Thank you for this code! I can't connect to the ECU. I tried L9637D and BC547. Esp32 connects perfectly to the browser, but there is no initialization. Any ideas?

muki01, 2025-03-07:
Hello. Thank you for the feedback. Are you sure your car has ISO14230 protocol? In the code, the default protocol is Auto, when the protocol is set to Auto sometimes it becomes a problem. If you have not changed the protocol in the code and if you are sure that your car is with ISO14230 protocol, you can change the protocol to ISO14230 and try.

1. The ignition of the car must be switched off.  
2. Connect the board you made and wait a little.  
3. Turn on the car's ignition.

vladinen, 2025-03-07:
I changed the protocols. 
Can you give me the simplest code for initialization ISO14230 fast
for single temperature parameter?

    digitalWrite(K_line_TX, HIGH), delay(3000);
    digitalWrite(K_line_TX, LOW), delay(25);
    digitalWrite(K_line_TX, HIGH), delay(25);

muki01, 2025-03-07:
Okay this code is only for esp32 and uses iso14230 protocol and takes coolant temperature. You need to use pin 25 and 26 for RX, TX and you can open the serial monitor to see what is happening. If you get hex data and then it says "Your Protocol is ISO14230_Fast" then your protocol is selected correctly and everything should work fine. My car also uses ISO14230 and with this code it works perfectly.

[Basic_Code.zip](https://github.com/user-attachments/files/19134416/Basic_Code.zip)

muki01, 2025-03-07:
This is a very short version of the code, it should work, but I won't be able to check in my car right now if it works, I can try tomorrow morning. 

[Basic_Code_short.zip](https://github.com/user-attachments/files/19134655/Basic_Code_short.zip)

vladinen, 2025-03-08:
Thank you for your reply. I'll try it today and let you know.

muki01, 2025-03-08:
Do you see any HEX data in the serial monitor or not?

vladinen, 2025-03-08:
Yes. i see 

[image]

muki01, 2025-03-09:
Why are you using Serial2? In my code I used serial1 with pins 25 and 26 because with serial2 it didn't work for me and I was getting the same data as you. Please use:
```
#define K_Serial Serial1
#define K_line_RX 25
#define K_line_TX 26
```

In the diagram connect the RX to GPIO25 and TX to GPIO26.

vladinen, 2025-03-09:
I've tried all the options, unfortunately it doesn't work.
my esp32
https://randomnerdtutorials.com/esp32-uart-communication-serial-arduino/
but, obd adapter easy connect to my car)

[image]

muki01, 2025-03-09:
Use this code. [Basic_Code.zip](https://github.com/user-attachments/files/19134416/Basic_Code.zip)
Don't change anything. 
In the diagram connect the RX to GPIO25 and TX to GPIO26 and show me the serial monitor.

vladinen, 2025-03-10:
Okay, I'll give it a try. I don't have a laptop, so I'll add the screen code to visualize the process. I installed L9637D instead of transistors, and it looks like the chip responds to the RX HEX code.

[image]

muki01, 2025-03-10:
Friend please first try my code that I gave you without changing it and without adding your codes for the screen and send me the picture. In order for me to help you you have to listen to me. First try only with my code without adding additional codes!

muki01, 2025-03-10:
You can connect the ESP32 to the phone with an OTG cable and open the serial monitor using the Serial USB Terminal program. It is available in the Play Store and the App Store.

vladinen, 2025-03-11:
Okay, I'll do it)

muki01, 2025-03-11:
It doesn't show you because you connected it wrong.  the black cable is the OTG cable it should be connected to the phone not to the ESP32 but I see your OTG cable is micro USB.  Do you have a Type C OTG cable or a Micro USB to Type C adapter?  If you don't have it, it won't work, you'll have to work with a laptop.

vladinen, 2025-03-12:
Thank you for your patience with me! I will try to find a laptop, that would be the best thing.

mgritsan, 2025-09-02:
Hello.
I have a 2007 Hyundai. ISO 9141-2 protocol (if I'm not mistaken)
The board is an Arduino nano clone. I use the code from your library examples, as well as AltSoftSerial.(Rx 8, Tx 9)
The circuit with the L9637 chip.
Please disclose the details of my errors. For example, the speed data is "-1". And OBD2 timeout

[image]

muki01, 2025-09-02:
> Hello.
> I have a 2007 Hyundai. ISO 9141-2 protocol (if I'm not mistaken)
> The board is an Arduino nano clone. I use the code from your library examples, as well as AltSoftSerial.(Rx 8, Tx 9)
> The [...]

Hello,
I think I understand your issue. There is a small problem in my code: if the protocol is set to Automatic, this issue may occur. I will make some adjustments to fix this in the code, but in the meantime, could you please try selecting the protocol as ISO9141 instead? That should work without any problems.

muki01, 2025-09-02:
> Hello.
> I have a 2007 Hyundai. ISO 9141-2 protocol (if I'm not mistaken)
> The board is an Arduino nano clone. I use the code from your library examples, as well as AltSoftSerial.(Rx 8, Tx 9)
> The [...]

If you’d like, you can also try using the Arduino library I created: [OBD2_KLine_Library](https://github.com/muki01/OBD2_KLine_Library).  
You can also find it directly in the Arduino IDE under the name **OBD2 K-Line**.  

⚠️ However, there might be the same issue when using the **Automatic** protocol.  
Please try selecting a specific protocol (e.g., **ISO9141**) instead.  

I will fix this issue in the code as soon as possible.

mgritsan, 2025-09-03:
I tried to select only one protocol. But as I understand, the error persists. Do I need to increase the timeout?

[image]

muki01, 2025-09-03:
> I tried to select only one protocol. But as I understand, the error persists. Do I need to increase the timeout?
> 

The data being sent looks correct, but it seems you’re not able to receive the values.  
Could you please try the **GetSupportedPIDs** example code, again using the **ISO9141** protocol, to see if it can retrieve data and which PIDs are supported?  

Also, does your OBD2 socket have pins **6** and **14**?  
Sometimes vehicles support both protocols but use **CAN Bus** as the main communication protocol.

mgritsan, 2025-09-03:
In my car, there is definitely no CAN bus. I tried the library from https://github.com/iwanders, and it shows me data, but only once with each new connection. And that library is complicated for my understanding. I will definitely try the example GetSupportedPIDs.

muki01, 2025-09-03:
> I found a small mistake in the example. It's just a typo. `13 Line.setDebug(Serial); // Optional: enable debug output on your chosen serial port`

Could you please try with the code in the file I attached?  
I’ve made a few changes to see what happens now. Normally, I was able to successfully retrieve data via **ISO9141** using this code, but on your side something seems to be going wrong. With this updated code, we’ll try to figure out what’s happening.  

📂 [Download the test code here](https://github.com/user-attachments/files/22126876/OBD2_K-Line.zip)  

You’ll need to extract the ZIP file into the folder where your Arduino libraries are located, replace the old library, and then run the **LiveData** example again.

mgritsan, 2025-09-04:
Hyundai Accent (2000 - 2010) engine - g4ec, petrol 1.5 - 102 hp.
Tried your new files:
[image]

mgritsan, 2025-09-04:
Just for information
My car has a non-standard Multitronics vg 1031upl on-board computer. During the tests, I disconnect it from the OBD connector, of course.
Here is what is indicated in the instructions for its protocols:
"The MK also supports universal diagnostic protocols in the “OBD2Q”, “OBD2P”,
“OBD2S”, “CAN”, “J1850”, , “Iveco” modes. The “OBD2Q”, “OBD2S”, “OBD2P” modes correspond to the OBD-II standard: OBD2Q - fast requests. “OBD2P”, “OBD2S” - slow"

In my case, OBD2Q was selected. This on-board computer works well.