muki01/OBD2_K-line_Reader #18: K-line without CAN

fmimolu, 2025-03-15:
A great project! I've been thinking about how to display data from K-Line for days, and I found your gem.
I have a Seat Ibiza Cupra 1.9 TDI that uses the ISO 15765-4 (CAN) diagnostic protocol for OBD2 communication.
The problem is that it's a DRAG project, and most of the wiring, including the CAN network, has been removed, so I'm only left with K-Line (ISO 9141-2).
I want to use a Dwin display to display some basic data (RPM, KM/h, temperatures, and a little more). The idea is to use Arduino Mega since it has UARTs available via hardware, and your schematic uses the L9637D.
Is it possible that your code will work with the CAN network removed?
Any suggestions for this?
Congratulations on such great work.

muki01, 2025-03-15:
First of all, thank you for your kind words! I haven't tested this project on a vehicle with two different protocols, so I can't be 100% sure, but theoretically, it should work. I have tested this project on vehicles with a single protocol, and it works perfectly for me. However, communication over ISO 9141 is quite slow, allowing only about 2 data requests per second. So, the data will be updated a bit slowly. Also, since the Arduino Mega has multiple hardware serial ports, it is a great option for communicating with both the vehicle and the DWIN display.

If you want, you can test the code, and if you encounter any errors or issues, I’d be happy to help!

fmimolu, 2025-03-15:
Thanks for your kind words and the offer of help.
I'm going to try it out. I've already ordered the L9637D to make the OBD-to-Arduino cable. As soon as I have it done, I'll run the initial tests.
First, I'll verify the protocol with the VAG group's VCDS to be completely sure which one it uses.
While the L9637D arrives, I'll adapt your code for the Arduino Mega.
Regards
P.D  Excuse my English...it's Google Translate

fmimolu, 2025-03-16:
Hi muki01, I tried VCDS today, and the protocol it uses is KW1281. Is it possible to use your code and schematics for what I want to do?
If you'd like to use another method to communicate, please let us know. This isn't a question-and-answer session, but I'll summarize and provide the results later.