muki01/OBD2_K-line_Reader #24: Question about schematic wiring – 12V_Pin16 connection

N0edL, 2025-05-20:
Hi, I was trying to recreate the schematic, but I’m unsure about the correct wiring.
Do I need to connect _12V_Pin16_ to both _GND_ and the _K-Line_Pin_, or only to _GND_?

#
[image]

muki01, 2025-05-20:
Hello. I'm glad to receive your feedback. On the diagram, please look for the red dots—only those indicate connected links. Therefore, in response to your question, the answer is "no," they are not connected.

N0edL, 2025-05-21:
> Hello. I'm glad to receive your feedback. On the diagram, please look for the red dots—only those indicate connected links. Therefore, in response to your question, the answer is "no," they are not connected. The 12V after the capacitor is not connected to the K-Line.

Ahh okay! One more thing. I’m trying to connect my ESP32-C3 to the circuit in a PCB design (it’s actually my first ever). I want to power the ESP directly from the 12V input. What’s the best component to step it down to 3.3V? If you have a small schematic for that, it would be really helpful. Thanks!